rbei-etas/busmaster #1174: BUSMASTER V3.2.2  - Kvaser Leaf Lin 2.0

Stefanuccio, 2018-05-24:
Hello , 

I've tried to configure busmuster tool in order to read LIN traffic with 2.0 protocol . 
The tool is able to reconize the hardware and the drivers  of the Kvaser Leaf , with the channel correctly configured as Slave node  ,but on the message windows appears only one frame "LIN message" with randomic update time and no ID ( despite correct LDF is loaded) .

Seems to be a problem of compatibility ... is there a way to use this hardware on busmaster ?

Thank you in advance

RBEI-busmaster, 2018-06-08:
Is the behavior same with other Kvaser LIN hardware as well ?

jalalkhan0, 2018-11-28:
I am facing the same issue with v3.2.2. Is there any suggest solution for this problem ?