rbei-etas/busmaster #1175: Inexpected close with ETAS 581.4

joelcote, 2018-05-25:
Hi,

I downloaded 3.2.2 to see if a fix on that problem was there but I still got the problem. I mostly use the Ixaat card on my laptop, But when I need to use it with another tool and I want to check the can I use my ES581.4 card. I installed the 3.2.2 and it work the first time. But after that  I select the right card in busmaster and each time I click on connect, busmaster close and a bug file is generated. (even if I close all my application) Do you know why this happen?

RBEI-busmaster, 2018-06-08:
Please share the dump file

joelcote, 2018-06-08:
I tried a couple of time this morning and I am not able to recreate the problem. Where Can I see old dump file that I generated in the past?

RBEI-busmaster, 2018-07-06:
Dump files will be created in the temp folder

joelcote, 2018-07-06:
Hi,

Are you talking about the " C:\Temp" folder? If yes what would be the name
of the file or sub folder??

-- 
<http://www.brp.com/>

*Joel Cote*
Ingénieur électrique,  3WV
Electrical Professional Engineer , 3WV
Mécatronique
565 de la Montagne | Valcourt | Qc | Canada | J0E 2L0 1-450-532--5100 #4857
brp.com <http://www.brp.com/>

The contents of this e-mail are confidential with all rights reserved to
the author. It is illegal to use or divulge this information without
authorization. If you have received this e-mail by mistake, please notify
us immediately by replying to the e-mail. Thank you.