rbei-etas/busmaster #1190: Screen Resolution Fix Needed

deo2016, 2018-10-10:
Hello,
The Message Window screen resolution is very poor. Even after using the magnifier tool on Windows, it is hard to read the messages. 
Kindly provide a fix to this issue. Please find image below. I have also tried installing all versions of BusMaster since 2.6.3 hoping the older format would not face the same issue.  
[image]

Teddyz, 2018-11-02:
[image]
I do not have the same problems.

digidax, 2018-11-09:
Same Problem like @deo2016 on Windows 10, poor resolution. The same on signal wazch windows. This size should be scalable by the user. During drivin a car to zomme the size of physical and raw value.
Thanks.

deo2016, 2018-11-09:
I am using a monitor with 1920 x 1080 resolution. That is where I am having issues. My older monitor was 1366 x 768 which did not cause any problems. I tried adjusting the resolution on my current monitor, but that is not solving the issue. I have simply started logging the CAN data and then later reviewing it. It is easier than straining my eyes.

digidax, 2018-11-09:
Yes, using als 1920 x 1080 with this problem.