rbei-etas/busmaster #1195: Node simulation issue (busmaster v3.2.2)

excision10, 2018-11-09:
I have a problem with the node simulation of busmaster. I follow the normal procedure for the generation of a new CAN node:

1. I've written the node in C in the editor of busmaster
2. I've built the node (the built generates successfully the .dll (but it does not generate the .def)
3. Since the .def was not generated automatically, I write by myself the DEF file.
4. Click on "connect", but at this point the error: "Unable to initialise handler functions for execution. The DLL  ......... is unloaded".

But, I do not understand how I am supposed to load this DLL. I am not able to find any "load" button. It does not exist! I already tried to build the node after the (manually) creation of the DEF file, and the build is always ok, but once I click on CONNECT, the same issue appears.

jogo-, 2018-11-09:
You can add another DLL in Node Simulation :

1. Add a Node
2. Enter the Node Name
3. Click Browse to select the DLL file

excision10, 2018-11-17:
Maybe I explained wrong, but the problem is that the DLL seems loaded correctly, but when I click on connect the mentioned error appears. It has to be a bug of BUSMASTER.

excision10, 2018-11-17:
The whole thing seems that can be solved by simply creating the node simulation files in a folder path not too long, as C:\Projects

closed 2018-11-17 by excision10