rbei-etas/busmaster #1199: busmaster 3.2.2 use ES582.1

wenluoma, 2018-12-13:
i have install ES582.1 driver, and i can see this device driver in device manager of PC.
but i cannot find ES582.1 channel in busmaster 3.2.2.
could you help me to solve this?
thanks.

wenluo

bahane, 2019-02-15:
Hi @wenluoma, 
I have exact the same issue, did you found a solution for it yet?
Newest driver and firmware is installed via HSP Update Tool.

KishoreComstar, 2019-08-19:
Hi All,
i am facing the same issue. Please help us if you have any solutions.
Many Thanks,
Warm Regards,
Kishore

ThirteenLLB, 2023-09-14:
i am facing the same issue.

PanPanchench, 2024-11-17:
Hi All,
i am facing the same issue.

PanPanchench, 2024-11-17:
Try to install BOA x86 package: https://etas.services/zh/downloadcenter/18102-41483.php
Thank you @sskaje