rbei-etas/busmaster #1206: Can't edit a CPP in node simulation

Stoneiron, 2019-02-11:
I'm trying to simulate a CAN network and I'm carefully following the guide, but I'm stuck here. It refers to a right pane that I can't see and I'm unable to find a working .cpp example. I tried to use as a .cpp input some examples found here on github, but Busmaster refuses to import it, because it says that they are not made for CAN. Could you help me?

[image]

jogo-, 2019-02-12:
For a working CPP example, just look at the Busmaster.zip here #1129

Stoneiron, 2019-02-12:
> 
> 
> For a working CPP example, just look at the Busmaster.zip here #1129

Thank you very much! Now I need some extra time to get deep into the syntax, but at least I have a working CPP file :-)

jogo-, 2019-02-13:
It's not so complex. I reproduced just the needed part of an ECM SW for a cylinder cutout test simulation.