rbei-etas/busmaster #1212: Sending Waveform Signals

duperoumat, 2019-03-13:
Hi,
I'm trying to debug a CAN Bus and I'd like to send a changing message on the bus (I'm trying to display a changing speed on a cluster).
I can send one message with no issues, but when I configure and enable a waveform, nothing is sent.
I configured the waveform, chose the right address from the database, started the connection, and enabled the waveform, but nothing happened.
Here is what the waveform looks like:
[image]
Here is what BUSMASTER looks like:
[image]
The 0x123 message is sent by the cluster to tell it's on and allows me to make sure it is connected.

Thanks for your help!

FrankGerlachFlex, 2019-07-31:
I successfully sent a message with a changing signal using Node Simulation. You can send ANY Signal your pice of C++ code can realize.
E.g:
//warning, untested code, please do a dry run on the console.
//in the 10ms Timer Handler:
    static time = 0;
    short y = (int16_t) (sin(time*2*3.1415f/360.0f) * 100);
     STCAN_MSG msg;
     msg.id = 0x18EF70C9;
     msg.isExtended = true;
     msg.isRtr = false;
     msg.dlc = 8;
     msg.data[0] = 0;
     msg.data[1] = 0;
     msg.data[2] = 0;
     msg.data[3] = 0;
     msg.data[4] = y ;
     msg.data[5] = y >> 8;
     msg.data[6] = 0;
     msg.data[7] = 0;
     msg.cluster = 1;
     time++;
     SendMsg(msg);

FrankGerlachFlex, 2019-07-31:
test123