rbei-etas/busmaster #1215: how to add a new driver to BUSMASTER

JieGuili, 2019-04-22:
I have a new CAN hardware and DLL file, I have put the DLL file in the BUSMASTER directory, but there are no options in Driver Selection as well, what should I do except putting DLL file in the install directory?

daniel-bostrom, 2019-05-02:
You need to update the BusmasterDriverInterface project located in the BusmasterKernel solution. Search for one of the existing CAN drivers (for example Kvaser) and add your new driver in the same way. I did it a couple of weeks ago and it works well.

limingruisss, 2019-06-06:
Hello daniel-bostrom:
I want to know how to generate CMakeLists.txt, And how to generate a build.bat

freeyyc, 2020-04-14:
> eeks ago and it works

build the BusmasterKernel solution can generate a dll file,then copy the dll file to Busmaster directoty. is this ok?

PrakashKumar0143, 2021-05-30:
Hi daniel-bostrom:
I am designing my own CAN hardware and trying to use this BUS Master Tool.
I have changed some other driver files like others how they did.
But I want to know the .dll file like how the PEAK CAN_PEAK_USB.dll is having.
Could you please help me with how to create this .dll file and what functionality is there & how I can make this .dll file so that my hardware will support by this BUSMASTER Tool.

Shubham260202, 2024-05-20:
Same Question, as @PrakashKumar0143  I have searched in the Git files but didn't find it, is there any solution for it

rusoku, 2024-05-20:
Just add  your project with your hardware support.
It is my driver version for a Rusoku TouCAN adapter.You can use it like reference design.
https://github.com/rusoku/busmaster_rusoku_driver

Shubham260202, 2024-05-23:
I saw the file but i didn't get you point. That how my hardware will be supported in busmaster. can you elaborate with points