rbei-etas/busmaster #1216: Busmaster development discontinued ???

mibcat, 2019-04-26:
After a longer time of absence I'm surprised about that no commit activities had taken place since Oct 2017. There are still some important issues (such as supporting Linux ;-)) ...

Ravenstone0910, 2019-05-07:
I guess supporting Linux won't come because this is a complete visual studio project on windows. As I know the develpment is stopped, but maybe one final release will be released.

rusoku, 2019-05-07:
Bad news then.I just started working on TouCAN converter driver for Busmaster...

FrankGerlachFlex, 2019-07-31:
@mibcat: The auto world does not like FOSS too much :-( 

But you could actually use the source code to create your own fork of BUSMASTER, which would run on Linux...

Or: maybe you get it running with WINE ?

Siogian, 2020-06-27:
this is a powerful software, firstly thanks the author's pay for, but the development cant be stopped, if the author is really disconnnect, who should fork this and i can join and develop it continue with each other, my english is not well, hope someone can hear me.

FrankGerlachFlex, 2020-06-29:
Hello, just read your message.

Unfortunately, BUSMASTER development seems to have stopped.

Forking it would be a great idea, but you would probably be mostly on your own.

Please let us know if you set up a fork.

Kind regards

Frank Gerlach

Siogian, 2021-05-10:
Hello, my fork has been created in https://github.com/Siogian/busmaster, Even a successful compilation is not easy，just build success using visual studio 2019，There is still a long way to go，Welcome to join us。

yegorich, 2021-05-11:
@Siogian great to see some movement in the project! What is your main purpose? To  make it compilable under Visual Studio or CMake?

teoCLMR, 2026-11-02:
For anyone still looking: I maintain a fork with regular releases, BUSMASTER-NG
(https://github.com/teoCLMR/busmaster-ng). It builds with VS 2022, runs on Windows 7–11 and fixes many
DBC-related issues. Windows only for now, no Linux support yet, sorry.