rbei-etas/busmaster #1218: Listing hardware interfaces failed

msd01, 2019-05-06:
Selecting any device results in an error message in the Trace Window "Listing hardware interfaces failed"

I jave tried ACNalyser XL and ES581.4 devices. 

The program used to work on this PC, but no longer.

I have tried a fresh instal and also made sure the tdm-gcc-4.8.1 components are correctly installed.

What else can be the cause of this problem???

gxurma, 2019-05-08:
seemss to be a windos 10 issue. have the same problem at fresh installs on fresh win 10. No solution yet...

msd01, 2019-05-08:
Could well be. I havent used BM since win 10 install, but it always worked on Win7

mhkr007, 2019-06-17:
> Selecting any device results in an error message in the Trace Window "Listing hardware interfaces failed"
>
> I jave tried ACNalyser XL and ES581.4 devices.
>
> The program used to work on this PC, but no [...]

hello msd01,
What else is required to connect busmaster to any hardware device, did we need any adapter or converter required to use busmaster to hardware interface.could you please give some information about connecting,any install required and how you connect to busmaster.

Newdea, 2019-12-28:
but i met the same problem on win7, when i tried to connect intrepidCS neoVI