rbei-etas/busmaster #1219: IXXAT with VCI driver 4.0

pagit2018, 2019-05-07:
VCI drivers 4.0 have been released for compatibility with Windows10 with SecureBoot PC, but the USB-to-CAN II device is not anymore detected by BusMaster (driver selection failed).
Everything was working fine with VCI 3.0 drivers.
Any suggestions?

IXXAT-wucherer, 2019-05-07:
Hi Pagit,

there is a new DLL wich supports IXXAT VCI V3 and V4. Also multiple controllers are supported now.
You can replace the DLL in the Busmaster installation directory and then it should work.

Issue:
https://github.com/rbei-etas/busmaster/issues/1191

Pull request here:
https://github.com/rbei-etas/busmaster/pull/1189

Best regards,
Peter

pagit2018, 2019-05-07:
Hi Peter,

thanks so much for your so fast and usefull reply.
I can confirm that updated DLL solve the problem and BusMaster works perfectly with VCI 4.0.

Sorry for opening this without searching before...

Kind regards
Paolo

closed 2019-05-07 by pagit2018

gxurma, 2019-05-08:
where do I get that DLL from?

IXXAT-wucherer, 2019-05-08:
[CAN_IXXAT_VCI_VCIV4_Support.zip](https://github.com/rbei-etas/busmaster/files/3155933/CAN_IXXAT_VCI_VCIV4_Support.zip)

IXXAT-wucherer, 2019-05-08:
Hi Martin,

I uploaded it here above.

The sources you can download here:  https://github.com/rbei-etas/busmaster/pull/1189

Please feel free to ask if there are problems or questions.

Best regards,
Peter