rbei-etas/busmaster #1220: Facing linking problem on Windows 10

rwaghode, 2019-05-15:
I have installed BusMaster V3.2 on windows 10. I have compiled MU5T.cpp file using TDM-GCC-64 and  to compile MinGW. I am getting linking errors. Please find attached error snapshot.

[image]

I also installed Busmaster 3.2 on windows 8 and the same cpp file is complied and created dll.

Is there any issue to run BusMaster v3.2 on windows 10.:

Please help...

jogo-, 2019-05-20:
Are you really trying to compile using TDM-GCC-64 and not TDM-GCC-32 ?

rwaghode, 2019-05-20:
I am using TDM-GCC-64

jogo-, 2019-05-22:
TDM-GCC-64 is not supported, see previous issues : https://github.com/rbei-etas/busmaster/issues?utf8=✓&q=TDM-GCC
You should uninstall it then reinstall it properly.
> For installation steps, please refer BusMaster User Manual 'Additional Installation->MinGW Installation using TDM-GCC Installer' available in the link "https://raw.githubusercontent.com/rbei-etas/busmaster-documents/master/help.pdf"