rbei-etas/busmaster #1222: how to connect busmaster to imx6 board

mhkr007, 2019-05-30:
Can you please help me busmaster to connect with imx6 board and in the driver selection which option i have to select..
[image]

yegorich, 2019-05-30:
What do you want to achieve? Do you have a CAN program running on imx6 and you want to debug CAN communication or do you want to use imx6 board as a CAN-to-PC converter?

For the first case you need a CAN interface. It can be a PCI-to-CAN card, USB-to-CAN or network-to-CAN converter. Such interfaces are well supported by BUSMASTER (PEAK USB, Kvaser CAN or VScom CAN-API to name few).

mhkr007, 2019-06-17:
> What do you want to achieve? Do you have a CAN program running on imx6 and you want to debug CAN communication or do you want to use imx6 board as a CAN-to-PC converter?
>
> For the first case you need [...]

Thankyou yegorich,I want to CAN programming on iMX6 quad processor board by using Busmaster.Is there any interface between imx6 and busmaster to work can controller.Can you please tell.

yegorich, 2019-06-17:
What operating system are you using on imx6? Linux?

mhkr007, 2019-06-18:
Hello Yegorich ,
    We used Digi.com Yocto project to build an image for the dash board in Linux PC and want to interface and control various peripherals with imx6 using CAN.