rbei-etas/busmaster #1230: Cannot install MinGw

toowaki, 2019-07-06:
I understand that Bustmaster need MinGw.
However, because of company security, MinGw can not be installed.

Busmaster Installation will stop at the screen which window is "Download update manifest. Downloading 'http://tdragon.net/tminst/net-manifest.txt'.".
If it is software that companies use, I think Busmaster will have to include MinGw in the tool to avoid such problems.
[image]

FrankGerlachFlex, 2019-07-31:
Did you try to delete the folder created by the installer ? I had the problem that it would not install after a partial package download. So I deleted everything and it worked.

toowaki, 2019-08-10:
Thank you for your reply.

Unfortunately, I cannot download MinGw at my office due to security.  So, the problem has not been solved yet.

But, busmaster seems to work without MinGw.
What function does busmaster use MinGw for?

FrankGerlachFlex, 2019-08-12:
If you just want to monitor the CAN bus, I guess you will not need Mingw and the gcc/g++ compiler. 
But if you want to write your own Rest Bus Simulation (with emission, processing etc. of CAN messages) , you will need mingw to compile your C++ Simulation Code.

You should talk to your corporate security for an Exception. Or bring your personal laptop and do all the BUSMASTER work on that laptop.

toowaki, 2019-08-16:
I understand what MinGw is used for.
I'll consult with my IT staff to install MinGw.

closed 2019-08-16 by toowaki