rbei-etas/busmaster #1233: ETAS ES582.1 not supported in Busmaster V3.2.2

KishoreComstar, 2019-08-16:
Dear Team,
i have recently purchased ETAS ES 582.1 and trying to use BUSMASTER 3.2.2 sW. Installed the drivers for this HW from HSP and windows recognized it. But BUSMASTER reports driver selection failed.
Please help with the correct version of BUSMASTER which will work with this HW.
Thanks,
Kishore

qzfx, 2022-07-12:
Did you install BOA x86?

ThirteenLLB, 2023-09-12:
> 
i meet same problem as  Kishore
sorry i do no understand , what is BOA x86

sabariraj33, 2023-12-19:
I am also facing the same issue, not able to select ETAS ES582.1 in BUS MASTER software. 
Installed HSP Package -> Installed ES582.1 Hardware driver from HSP tool. 
BOA Distribution package v2.35.0.57 

Still getting driver selection failed message from BUS MASTER

ThirteenLLB, 2023-12-19:
Is the BOA you installed the x86 version, if you install the x86 version of BOA and es582.1 at the same time and still can't recognize the driver, then I don't know what to do

 

 Installed HSP Package -&gt; Installed ES582.1 Hardware driver from HSP tool.