rbei-etas/busmaster #1237: Interfacing Busmaster with CAN module like mcp2515 and arduino

krish97git, 2019-09-26:
Hey, anyone tried to interface the busmaster with CAN module like mcp2515 and arduino or anything similar?
If so what is the procedure?

yegorich, 2019-09-26:
You'll need for example a USB-to-CAN adapter attached to mcp2515. This way you can communincate with Arduino via a CAN bus.

krish97git, 2019-09-26:
Don't I need specific drivers for BUSMASTER to successfully run it using arduino and MCP2515
I dint get any drivers for that, so arises the question.

yegorich, 2019-09-26:
AFAIK you cannot attach Arduino directly to BUSMASTER. So you if have an application that is using CAN on Arduino and Arduino interfaces with CAN through MCP2515 (SPI bus) and you want to use BUSMASTER to sniff on the CAN bus then you need an interface already supported by BUSMASTER. Like for example this [one](http://www.visionsystems.de/produkte/usb-can-plus-usb-can-plus-iso.html).

viperidae, 2020-09-07:
Is there any plans for adding support to connect to serial devices that use the slcan interface?
https://github.com/linux-can/can-utils
https://elinux.org/Bringing_CAN_interface_up#SLCAN_based_Interfaces

It's relatively straight forward to implement slcan on an arduino