rbei-etas/busmaster #1246: CAN diagnostic: Request diagnostic service difficult

EricXiaoke, 2019-11-28:
We always have to set the CAN diagnostic configuration.
Then send diagnostic.

I think if we can do as below, it will be powerful.
1. write the diagnostic service to excel according to specified format
2. busmaster loads the diagnostic excel
3. we can click the service like CANoe diagnostic console.
4. it can generate the auto test script like diva
Thanks.

alesiem, 2020-04-23:
I fully aggree, could someone fix that please?!

EricXiaoke, 2020-04-26:
I tried to add the functions to the busmaster compile project.
However, it compiled failed.
So we can add our tester.cpp to Node Simulation to implement the functions
1.load excel(csv) in tester.cpp
  implement it in visual studio first, after it is ok, move it to tester.cpp
2.develop CAN diagnostic message send and receive component
  implement it in visual studio first
    use FreeRTOS queue to simulate CAN message send and receive
    implement tester and server tp layer(FF SF FC CF)
    implement tester and server uds layer
    load the diagnostic excel, tester send queue to server queue
  move the visual studio code to tester.cpp by replace send and receive interface  with queue
3.develop WinApi
   develop WinApi in visual studio first, then move it to tester.cpp
4.develop diva test
   make the test algorithm in visual studio, then move it to tester.cpp