rbei-etas/busmaster #1252: Message Handlers in 3.2.2

Satheeshkk, 2020-02-10:
Hi,

I'm new to Busmaster. I'm working with v3.2.2. I want to monitor CAN data with particular IDs and give response automatically from script. I tried node simulation with message handlers, but it is not working. Script is building and creating .dll successfully. But script is not receiving and not transmitting anything. Function body is written in the function that is created when I add a message ID to message handler. But this function body is not saving in the .cpp file which is loaded when a node is created. I gone through the help document, but it is not helpful. 

ScreenShot1: Written script in .cpp file
ScreenShot2: Added message to message handler, that creates same function with empty body.
ScreenShot3: Commented the function which is written in .cpp file and added function body in new created function in pane 3.

[image]
[image]
[image]

Please help me in writing busmaster(v3.2.2) script which is to receive and respond automatically on particular ID.

Thanks.

mr52k, 2023-04-20:
Hii Satheeshkk, i am also facing same problem  with bus master v3.2.2 if you got the problem resolved then share the solution steps

mr52k, 2023-04-20:
+1 more if done with that peoblem plase help me in that thank you

mr52k, 2023-04-20:
Hii satheeshkk, in your script you need to use on bud connect, after that only you will get that.