rbei-etas/busmaster #1253: Busmaster does not run on my Windows10-64bit

Rikiluciani, 2020-02-13:
OS: windows 10 enterprise
System type: x64
Busmaster v3.2.2

After clicking on BUSMASTER icon to run it, BUSMASTER start up  but the program freezes and remains blocked. It cannot be closed (only way it is to force closing with Ctrl+Alt+Canc).

[image]

I tried to adjust the compatibility mode, but the issue was still there.

quantum8, 2020-03-31:
Might be some group policy issues, or something else clashing with the app on your computer?
It works on mine running Win10 version 1909.

rusoku, 2020-05-21:
It works for me.My win 64bit version 1909 (OS Build 18363.863).
But I recompiled BUSMASTER under Visual Studio 2019.

aheike, 2021-11-25:
Dear all,

Me and my colleagues are facing the same issue like described above. Some group policy issue sounds like a possible root cause, but is there any suggestion on how to further investigate or improve?

Is there any other support address, logfiles, ... ?

Thank you and best regards

gxurma, 2021-11-26:
is there actually anyone still maintaining and improving Busmaster?
Martin

quantum8, 2021-11-26:
There's been no commits for years, so no.
There was a fork created but I'm not sure how far it got.
https://github.com/Siogian/busmaster

> is there actually anyone still maintaining and improving Busmaster?

0p3n-s0urc3r3r, 2025-03-21:
I ran into this exact issue on a company computer.  I spoke with our IT department and they were able to fix this by installing it with their admin account.  I believe there was also an issue with group policies as mentioned before.

teoCLMR, 2026-11-02:
A likely cause of the freeze at start-up: the simulation engine `BusEmulation.exe` is a COM server that the
original installer registers with administrator rights. If that registration is missing (no admin rights,
group policies), BUSMASTER waits for it forever on the splash screen — which fits the comment above about
installing with an admin account.

In BUSMASTER-NG (a community fork, https://github.com/teoCLMR/busmaster-ng) 3.4.1 and later, the engine is
registered for the current user when needed, without admin rights, and if it still cannot start you get a
message that says why instead of a hang.