rbei-etas/busmaster #1254: Test Automation Executor does not send CAN messages with PEAK USB

nemomar, 2020-02-19:
BUSMASTER v3.2.2
Steps to reproduce:
-Setup a BUSMASTER connected to PEAK USB driver
-Associate with a .dbf (safetyPLC.DBF) where the can messages are defined
-Open Test Automation Editor with (safetyPLC.xml)
-Verify that DataBase Path points to the same safetyPLC.DBF
-Open Test Automation Executor
-Add safetyPLC.xml
-Execute
-No message is transmitted!
[safetyPLC.zip](https://github.com/rbei-etas/busmaster/files/4225951/safetyPLC.zip)

Sending messages through the Transmit Window works without problems. I tried all the channels available (from 1 to 16). What am I doing wrong?

alesiem, 2020-04-23:
I have the same problem with a Kvaser CAN!

kishorehos, 2022-03-25:
I too have the same problem with PEAK USB.

xzp11, 2023-02-07:
Maybe it is a path question.   you can move safetyPLC.xml and safetyPLC.DBF to C disk，and try again