rbei-etas/busmaster #1267: BUS MASTER Driver Issue

kushagra088, 2021-02-11:
I am facing an issue while connecting ETAS ES582.1 with BUS MASTER, showing the error "Device Selection Failed". A snapshot is attached for the same.  I already confirmed ES582.1 drivers installed on the PC. Hardware E582.1 is also working with INCA.
[image]
[image]

YashasNanjundagupta, 2022-03-08:
I'm facing the same issue, is there any solution given already?

YashasNanjundagupta, 2022-07-13:
Hello Team,

No. I have not!

_____________________________________________________________________________

Best Regards,
Yashas Nanjundagupta

2 wheeler powersports (2WP/ENG2-IN)
Bosch Limited - (PAN AAACM 9840 P) - (CIN: L85110KA1951PLC000761) | Post Box No:3000
Hosur Road, Adugodi | Bengaluru-560030
Karnataka, India | INDIA | www.bosch.in
Tel: +91 9844608642 | ***@***.******@***.***>

Registered Office: Bosch Limited, Hosur Road, Bengaluru-560030, Karnataka, India
Managing Director: Soumitra Bhattacharya

Have you installed BOA x86?

-

sabariraj33, 2023-12-19:
I am also facing the same issue, not able to select ETAS ES582.1 in BUS MASTER software. 
Installed HSP Package -> Installed ES582.1 Hardware driver from HSP tool. 
Installed BOA Distribution package v2.35.0.57 

Still getting driver selection failed message from BUS MASTER