rbei-etas/busmaster #1268: COM API with python

ravikumarmedi, 2021-03-16:
Hi
i am using COM API function "SendCANMSg" in python and currently i am getting error 
"raise AttributeError("%s.%s" % (self._username_, attr))
AttributeError: CAN_MonitorApp.Application.SendCANMSg"

unfortunately i am not able to find proven python example.  here is my code.  can somebody help to figure out the issue
--------------------------------------------------------------------------

import ctypes as ctypes
import struct as struct

import win32com.client
import timev
BUSMASTER_COM_NAME = 'CAN_MonitorApp.Application'
myBusMaster =win32com.client.Dispatch(BUSMASTER_COM_NAME)
CONF_FILE = b"E:\PYthonprogramms\busmaster1.cfx"
#myBusMaster.LoadConfiguration(CONF_FILE)
myBusMaster.Connect(1)

CanDataFD = ctypes.c_uint8 * 64
class struct1(ctypes.Structure):
   _pack_ = True  # source:False
   _fields_=[
             ('m_unMsgID', ctypes.c_uint32),
             ('m_ucEXTENDED', ctypes.c_byte),
             ('m_ucRTR', ctypes.c_byte),
             ('m_ucDataLen', ctypes.c_byte),
             ('m_ucChannel', ctypes.c_byte),
             ('m_ucData', CanDataFD),
             ('m_bCANFD', ctypes.c_bool)                    
             ]

   def __init__(self):
         self.m_unMsgID=45678
         self.m_ucEXTENDED=1
         self.m_ucRTR=1
         self.m_ucDataLen=8
         self.m_ucChannel=1         
         for i in range(64):
             self.m_ucData[i] = 1             
         self.m_bCANFD = 1
         
sMsg=struct1()

#myBusMaster.SendCANMSg(sMsg)

zmicek, 2021-08-06:
I have similar issues. 
At first i had problem with LoadConfiguration() but somehow that function started working. I was passing path as:
CONF_FILE = u"D:\config_file.cfx"
and 
CONF_FILE = u"file://localhost/D:/config_file.cfx"
in order to pass it as BSTR.

Another issue is that I cannot start logging with StartLogging() function.
> BUSMASTER_COM_NAME = 'CAN_MonitorApp.Application'
> myBusMaster =win32com.client.Dispatch(BUSMASTER_COM_NAME)
> CONF_FILE = u"file://localhost/D:/config_file.cfx"
> myBusMaster.LoadConfiguration(CONF_FILE)
> myBusMaster.Connect(1)
> myBusMaster.StartLogging()

Error that I get is:

> In [37]: myBusMaster.StartLogging()
> ---------------------------------------------------------------------------
> AttributeError                            Traceback (most recent call last)
> <ipython-input-37-b7a50a6e68a0> in <module>
> ----> 1 myBusMaster.StartLogging()
> 
> c:\python38\lib\site-packages\win32com\client\dynamic.py in __getattr__(self, attr)
>        552
>        553                 # no where else to look.
> --> 554                 raise AttributeError("%s.%s" % (self._username_, attr))
>        555
>        556         def __setattr__(self, attr, value):
> 
> AttributeError: CAN_MonitorApp.Application.StartLogging

Does anyone knows is this function works correctly or I am doing something wrong?

zmicek, 2022-09-08:
Not that i know.