rbei-etas/busmaster #1270: Replay not respecting recording timings despite configuration

marinaalice, 2021-04-15:
Hello,
I have a recorded file from BUSMASTER that I want to replay.
The time stamps imply a certain duration, as well as a simulation video that recorded the session.
My replay is three time slower then expected.
Attached my configuration.
Thank you for any explanation you may provide.
[image]

Best regards,
Marina

TaekGu-Lee, 2022-01-03:
Hello.

I too have the same issue.
Logged messages are sent 3x slower on replays.
If you replay the 20ms cycle CAN message recorded with BUSMASTER, it will be transmitted every 60ms.

It is the same whether using PEAK CAN or running simulation mode.

thank you.

ex) Log file : period 20ms => Replay every 60ms.
*Log file
15:41:10:5179 Rx 1 0x387 s 6 00 00 00 00 00 00
15:41:10:5378 Rx 1 0x387 s 6 00 00 00 00 00 00

*Message Window
00: 00 : 00: 0568 Tx 1 s 0x387 WHL_PUL11 6 00 00 00 00 00 00
00: 00 : 00: 0625 Tx 1 s 0x387 WHL_PUL11 6 00 00 00 00 00 00

ghidoli, 2022-04-20:
Hello,
tested in 3.2.2 and in 2.6.0. Replay do not respect timestamp in "log" file.
In my case, in simulation mode, I got a 2.31 multiplier (in total duration time of an quite "low rate" log file)
Also using peak CAN USB I got similar results. 
I understand that windows is not a realtime OS... but I'm pretty sure it is possible to obtain more accurate replay time.