rbei-etas/busmaster #1292: Error in "Driver Selection" for ES582.1 using BUSMASTER 3.2.2

iscopa, 2022-11-04:
Hello,

I am using BUSMASTER version 3.2.2 and ESP582.1 Firmware v4.0.1.

When I click on "CAN->Driver Selection->ETAS ES582.1" the trace window prompts "Driver selection failed"

Is there a tested BUSMASTER version and ES582.1 firmware version you can recommend, please?

Thank you

woxkrs1, 2022-11-23:
Same problem here. Using Busmaster v3.2.2 and ES582.1 Firmware V4.0.1

gauravagnihotri, 2022-12-03:
This project doesn't seem to have received any attention since 2017. It's really an amazing tool.

sskaje, 2023-01-03:
Try to install BOA x86 package: https://www.etas-cn.com/zh/downloadcenter/18102.php

woxkrs1, 2023-01-12:
The BOA x86 package working out just fine! It is now possible to select the ES852.1 driver without any error.
Thank you @sskaje for your help!

CodeWritingAcarid, 2023-05-04:
Thankyou for help! GOD BLESS YOU!