rbei-etas/busmaster #1299: BUSMASTER Activity

autologicbmw, 2023-06-05:
Is Busmaster still live or is it over?
Saravanan are you still with us?

vguttmann, 2023-06-18:
The last edit was 5 years ago, so yeah, I'd assume it's dead.
There are no active forks either, so anyone that reads this and is looking for a project to adopt can do so.

saerpa, 2023-07-12:
Any recommendation?

autologicbmw, 2025-08-08:
> Hi, our company based in Germany is going most likely to start fixing some of the issues we have with Busmaster, so we are looking at this possibility, but we are not sure should we also upstream our changes here. But since this project has nobody to merge the pull requests, is there any point to upstream? License is GPL, so that states that any changes made should be contributed back here. Unless I misunderstood.

I’m very interested in this

 

 

Ian Crane  BSc RSV

BMW Specialist

Australia

0409578575

***@***.*** ***@***.***>

reymor, 2025-08-08:
That was the idea. Trying to support cross-platform and I thought that maybe Qt was a good choice. It's true that I got into some problems like:

- The database manager was close source from version 3.x onwards and there are issues with that and I cannot solve them. I was trying to contact ETAS/Bosch/Old maintainers and I got answer from old maintainers but they don't have access to the source code anymore. So, here the only solution that I saw is to rewrite the component.
- There are other close source components like using the peakcan close drivers
- In general the source code is a bit messy because in some point they started to rewrite from a monolithic application to a more modular approach and it seems to me that they didn't have the time to finish the job.

So, in the meantime my goals changed like thinking: ok, maybe it is not a bad idea to rewrite from scratch or just to maintain in MFC and keep functionality. Also, UDS functionality is not clever and a bit messy.

In meantime, I discover this other project ([ecu-bus](https://github.com/ecubus/EcuBus-Pro)) which seems pretty cool and I was trying to maybe work together with that project, at the end both are trying to the same provide the functionality that is provided by expensive can tools. The 'problem' with that project is that is written in electron which i don't have experience with

autologicbmw, 2025-08-08:
Also display text in message window incompatible with some high resolution screens

 

Text in window too small

 

 

Ian Crane  BSc RSV

BMW Specialist

Australia

0409578575

***@***.*** ***@***.***>

docsassy, 2025-10-07:
I'm also looking at using Busmaster going forward and can help fix the issues in this thing.

Couple of quick questions:

What sort of support does BM have CAN-FD? 
Do the Intrepid drivers support the later ValueCAN's like ValueCAN4?

cheers,

Docsassy

autologicbmw, 2025-10-08:
> Hi,
> We have started to work on a new version of Busmaster with a different name already and it will be released sometime early next year. It will have a major UI rework (still MFC), plus some additional features that are for CAN/ARINC812A purposes that fit to our company's needs. But probably no changes to drivers since CAN drivers work fine for us already. But it will still be Busmaster under the hood, so it could be used as a base for a new version of Busmaster later if there is someone who wants to work on that.

Good news

 

 

Ian Crane  BSc RSV

BMW Specialist

Australia

0409578575

***@***.*** ***@***.***>

teoCLMR, 2026-11-02:
The original project has been quiet since 2017, but BUSMASTER is alive as a community fork:
BUSMASTER-NG, https://teoclmr.github.io/busmaster-ng/ (releases, installer, issue tracker).
Not affiliated with ETAS/Bosch.

autologicbmw, 2026-11-04:
Hello,

 

Tell me more.

 

Thank you

 

 

 

Ian Crane  BSc RSV

BMW Specialist

Australia

0409578575

***@***.*** ***@***.***>