rbei-etas/busmaster #1301: Added support TouCAN adapters from Rusoku technologies

rusoku, 2023-08-29:
driver supports up to eight TouCAN devices at the same time
https://www.rusoku.com/storage/app/media/drivers/Windows/BUSMASTER_RUSOKU_driver_v1.0.0.zip

[image]

gxurma, 2023-08-29:
wow. Stable?
Tested? For How long?

https://www.rusoku.com/storage/app/media/drivers/Windows/BUSMASTER_RUSOKU_driver_v1.0.0.zip (https://www.rusoku.com/storage/app/media/drivers/Windows/BUSMASTER_RUSOKU_driver_v1.0.0.zip)

	 (https://user-images.githubusercontent.com/46974254/264129913-9ad1c442-597f-436f-af3e-f9ae93c6ad10.PNG)

rusoku, 2023-09-23:
Released a driver v1.1.0 for the RUSOKU TouCAN adapters.
https://www.rusoku.com/storage/app/media/drivers/Windows/BUSMASTER_RUSOKU_driver_v1.1.0.zip

*added hardware timestaps support from adapter
*added error frames & statistics support
*small driver code improvements

rusoku, 2023-09-23:
BUSMASTER v.3.2.2 installer with the RUSOKU TouCAN adapters support:
https://www.rusoku.com/storage/app/media/software/Busmaster/BUSMASTER_Installer_Ver_3.2.2.zip
WIN SDK: 10.0.22621.0
VC++: 2022 v143