rbei-etas/busmaster #1305: Novo BUSMASTER

reymor, 2023-12-08:
Hi everyone,

I have forked the BUSMASTER project and I named **Novo** BUSMASTER. I truly believe in open source and I think that BUSMASTER is a really good idea. There are not many options in this field and I think that it is worthwhile trying to keep a project like this. 

The first thing that I did was to compile the documentation which I post here: 

[BUSMASTER documentation](https://reymor.github.io/documentation/2023/12/06/busmaster-docs.html)

The other thing that I did was to compile the project from source code, by now I think that I had the development environment. 

I would like to say also that I am not a GUI developer. In addition, my area of expertise is not exactly these kind of programs. However, I would like to start this project and to know see what happens. 

Trying to compile the documentation and the project itself I had the opportunity to read some code and trying to understand the real state of the project. My initial thoughts is that BUSMASTER is not a easy project as maintainer. I face a problem with the compilation process and it was kind of difficult trying to figure out where was the problem. 

As I said, I truly believe in the open source community. I think that the effort done in BUSMASTER was amazing and I would like to continue this project. I have some ideas to improve and add new features, some of them are difficult like porting to Linux. However, I will start fixing some bugs that were reported and in that way trying to understand more the project. This will take me time. I will try to read all the issues that are open here and in some cases try to fix them. If there is somebody which would like to join me in this adventure I would appreciate that. Also, if you think that there is any feature which would be something interesting to add just open an issue and I would try to address the implementation.

[Novo BUSMASTER](https://github.com/reymor/novobusmaster)

Best regards,

rusoku, 2023-12-09:
I spent a lot of time on the source code of this project. I developed several CAN and LIN drivers for the BUSMASTER program and TouCAN Rusoku adapters. And in my opinion it is better to make similar software again from zero but on more modern GUI framework base like Qt or similar.It also would work under windows, linux, macos.
IMHO

reymor, 2023-12-09:
Hello @rusoku 

Thanks for you feedback I really appreciate that. I saw that your post on these adapters. I would say that first I thought exactly the same. I mean I am not an expert on this things. However, I have spent time on really big projects like linux kernel, android source code, etc. When I saw the code and the dependencies I though maybe it is better to write this from scratch having in mind the cross-platform. 

As I said this was my first idea and I was not really sure about that. So, I just decide start this journey and to understand more the product. 

I think that the one of the first thing is trying to change the build system from Visual Studio to some cross platform build system like cmake. There are modules that seems to have that support. In addition, there are modules which are written using Qt. However, there are more written using MFC. 

The reason that I choose to migrate the project to some modern GUI instead of writing one from scratch is that this project seems to have some community. There are people that use this software like you point out that you add support for new hardware. With this in mind I thought that maybe makes sense trying to migrate the project but in the road we could try to fix some bugs and add new support for new hardware. The ideas behind the modular approach does not seem really odd and to me it seems that they started this migration. I saw this paper [BUSMASTER – An Open Source Tool](https://uwispace.sta.uwi.edu/server/api/core/bitstreams/aaf02ee1-bc1d-4046-9d51-152e0ffb9b9f/content) and it seems that there was progress in this direction of cross-platform, but it seems that they lost the support or maybe they conclude that it was better to write a new software from scratch and start selling XD.

I'm with you when you said that it seems better to start a project from scratch. However, I saw that I could start the migration from MFC to Qt and in the mid-time they can live together as it is right now. Regarding this migration, I saw the roadmap as: 

- Migrate from Visual Studio to some cross plataform build system (e.g cmake)
- Migration from MFC to Qt
- Clean the project to be able to compile in Linux/Windows

These are really difficult task. This are not easy things. The idea in the middle is trying to fix and add support to new hardware. I'm not really sure about this, and maybe I lost my path in the middle it is something that I'm going to see. Also, that's the reason that the feedback that you tell me it is really important for me and I'm trying to keep everything in my mind. 

If you are open to share your work, I will be happy to add and start creating the first realese of Novo BUSMASTER with the HW that you added support :)

gxurma, 2023-12-11:
Hi Guys!
As a user of Busmaster, hacking some CAN bus, I recommend following:

Start from scratch. too many hidden things that hinder correct function. For example, I always struggled with setting up the filtering functions. Sometimes it worked, manly not, sometimes it worked when I did not want it etc. 

Use Python. That´s usable on all platforms.
Today PyQt5 would be good to use. No need to deal with all these build craziness. The moment you decide to use any build system, there pop up 10 new and you are immediately outdated.
Designer for the Qt design, and any text editor for the python part. Visual code is good enough.

Use a hardware abstraction layer, where it gets to the communication to the drivers. That makes it extendable.

I recommend also writing the DBC editor from new.

I was happy to have Busmaster as a tool when i needed it, for example the node simulation was meant good, but gets strange when having more nodes.
I recommend to use here python too.

actually it is a pity that busmaster was abandoned since 2017.

Yours sincerely,
Martin

rusoku, 2023-12-11:
BUSMASTER driver source code for Rusoku USB CAN adapters win32
https://github.com/rusoku/busmaster_rusoku_driver

Qt5.15.x SerialBus can plugin source code for Rusoku USB CAN adapters (WINDOWS x64, MacOS)
https://github.com/rusoku/QtSerialBus

Qt6.6.x SerialBus can plugin is in developing stage.It will works with a new CAN FD, LIN device series from RUSOKU (USB3.x ETH, WiFi7)

In fact, I have started to develop a similar open source program based on Qt6 and C++ and I would like to support more protocols like CANopen, VSCP, D-PDU  etc...

reymor, 2024-01-18:
## Novo Bus Analyzer v0.0.1
* This first version is based on BUSMASTER v3.2.2
* Add `cmake` as build system - Tested project compilation with `vs2013` or `vs2019`. 
* Add support for TouCAN - Not tested
* Fix #5 - Peak PLIN USB does not work (one channel)

#### Note

This release was done using `cmake` as build system generator for `vs2013`. I recommend using `vs2019` as development. 

[Novo bus analyzer v0.0.1](https://github.com/reymor/novobusanalyzer/releases/tag/v0.0.1)