rbei-etas/busmaster #1306: Does BUSMASTER supports CANopen device "Absolute encoder from Piher Sensing Systems" ?

sandeepreddy0000, 2024-01-08:
I am trying to connect a CANopen device (Absolute encoder from Piher Sensing Systems) to BUSMASTER using kvaser Leaf Light hardware. However, I am unable to access the CAN message. Can someone please confirm whether BUSMASTER supports CANopen or not?

sandeepreddy0000, 2024-01-09:
Thank you for your reply.
Can you please suggest some open source softwares like BUSMASTER for CANopen ?

gxurma, 2024-01-09:
> 	Hello @sandeepreddy0000 (https://github.com/sandeepreddy0000)
>
> 	BUSMASTER does not support CANopen

??? Busmaster clearly supports CANopen :D !!!

	CANopen is just a kind of definition of what CAN message IDs belong to what information, and whats in the message.
So it definitely supports listening to and sending CANOpen messages too!
I extensively was using it last year for exaxtly that purpose. Talking to CANOpen based motor driving VFDs.

Bye,
Martin

reymor, 2024-01-09:
Well, I think that we have a different view of what CANOpen is. 

Having the message a buffer and doing a interpreation is NOT support CANOpen. This is exactly the same idea with UDS, you could say, you can make the payload with the requirements, etc, but it is NOT support for UDS. 

If we talk in the context of other open source tools, like Wireshark. You have an ethernet trace, and if you are using a propietary protocol, which is not supported by Wireshark, it does not mean that you are not able to see the data. You will see the data, but without any meaning, that what that propietary protocol means to you by that. You could offcourse create a dissector, etc. But clearly Wireshark does not support your propietary protocol. 

In this case, off course you can receive CAN messages, but without any meaning because Busmaster does NOT support CANopen. Could you interprent this data? Off course, you could but it does not mean that Busmaster support CANopen.

quantum8, 2024-01-16:
Is the error continuously being transmitted? I.E.: is the time for the error message incrementing, or is it once off. If once off then you should be able to clear the message window and it won't return.

If it does then you either have a hardware or software configuration issue.

A Bit error is:
```When the bit received is not the same as the bit transmitted. Every node that receives the data reads it bit-by-bit from the Bus and compares the transmitted bit value with the received. If the bit transmitted and the bit received are not of the same value, a "bit error" occurs.```

My first thought for hardware issue is perhaps you haven't terminated correctly. Is there termination at both ends of the bus? You need 120ohm at both ends, not just one end. Sometimes you can get away with no termination, or just one but that's usually on a short cable run at low transmission rates, and certainly not in production.

gxurma, 2024-01-16:
> Thank you for your suggestions.
> I connected the CANopen device to Busmaster using the Kavser leaf light - (120-ohm termination resistance is connected, and the correct baud rate is selected). When given NMT cmd to get into operational mode, I get an error with id:0x2001 shown in the image.
> IMG20240114143043.jpg (view on web) <https://github.com/rbei-etas/busmaster/assets/123000600/eb365809-1ae3-42f8-b688-5624c6e93049>
> While trying multiple times, I got an Rx message from the encoder shown in the image, but I am unable to get that again.
> IMG20240114151454.jpg (view on web) <https://github.com/rbei-etas/busmaster/assets/123000600/48c6d982-bd79-4675-b444-782f72920132>
> Can someone please suggest where I am making the mistake?

I  sometimes get similar errors, when i think that the device communicates with a for example  1Mbit/s but it really is only doing 500kbit/s.

Have a look at the lines with an oscilloscope, an check the bit speed there. Also the signal quality.

 

At least you already have at least 1 message seen 😃

 

Just also check the datasheet and manual of your device, that you want to talk to.

And set it first to a slow speed like 250k or 500k.

 

Bye,

Martin

www.enGYneer.at

sandeepreddy0000, 2024-02-28:
Thanks for your replies.
Now, I can read CAN output in BUSMASTER. I tried changing the baud rate and CANID of the CANopen encoder using the LSS service, but my device is not responding if I transmit using bus master. Can anyone suggest to me about this?

I just want to highlight that the encoder is giving output without any NMT command....

gxurma, 2024-02-28:
Hi!
If you need deep diving into your CAN project,
please contact me at ***@***.***
We might help.
Yours sincerely,
Martin Gyurko
enGYneer.at