rbei-etas/busmaster #1314: Uninstall Busmaster

FoxBzh49, 2024-04-30:
Good morning,
I cannot uninstall the Busmaster software automatically. The only argument that works is "/S", but the software does not disappear.

$UnInstallPath = "C:\Program Files (x86)\BUSMASTER_v3.2.2\uninst.exe"

$ArgumentUninstall = "/S"

tart-Process -verb runas -FilePath $UnInstallPath -ArgumentList $ArgumentUninstall -Wait

Do you have a solution please?

Thank you so much

gxurma, 2024-04-30:
Unfortunately, Never tried to uninstall Busmaster.
I only install it on new PCs and then it works as intended.
I do not think that it installs lots of bloatware etc, so I think uninstalling it through the windows app removal should work normally, as it is meant to do.
If you already tried to do that, and it did not work correctly, I think then something still was running and holding the folder, preventing the complete uninstall.
Maybe you can manually just remove it.
Maybe it is a Win11 issue. As most network problems nowadays.
Bye,
Martin Gyurko
enGYneer.at

FoxBzh49, 2024-04-30:
Good morning,
Thank you for your response and for taking the time to respond to me.
Actually, my question is about mass uninstallation. Indeed, I work in several educational establishments and we deploy our applications using SCCM. It is therefore necessary to have the possibility of massively uninstalling the product in order to upgrade to a new version.
The silent argument is in the documentation, but it doesn't work.
The publisher just needs to fix this problem. Manual uninstallation is not possible.