rbei-etas/busmaster #1317: Adding driver support for custom hardware

S-adnan556, 2024-08-20:
Hello, I’m working on adding support for a custom CAN hardware interface to Busmaster. So far, I’ve made UI changes by adding an enum for my driver, adding the driver name in the driver list in sg_ListDIL (Sources/Kernel/BusmasterDriverInterface/DIL_CAN.cpp), and loading the .dll in the DILC_SelectDriver() function. I successfully built and ran Busmaster by copying and renaming an existing .dll for my driver. My custom driver appears in the GUI, and if I connect it using the copied .dll (e.g., PEAK), it works for that hardware.

Now, I’ve copied and renamed the CAN_Vector_XL and included it in the project. I want to make changes to it so that it can work or my custom board. I need guidance on the next steps to enable communication with my custom hardware. Any documentation or pointers would be greatly appreciated, as I'm not sure how to proceed.

rusoku, 2024-08-20:
So now delete the Vector API and put your own

S-adnan556, 2024-08-20:
By Vector API do you mean replace this vector .dll with my own /Sources/BUSMASTER/BIN/Release/vxlapi.dll 

Right now our custom board is connected over a USB COM port

rusoku, 2024-08-20:
Take the Vector driver source code and replace it with one that works with your hardware.

sandeepcembed, 2024-08-20:
I am working on similar i have a device which works serial com port (NXP rt controller), so how can i integrate with BUS master?
Do you have any reference material? any example to integrate with Bus master.

I have implemented with Qt monitor its working fine with SL-can. 
https://github.com/tixiv/CAN-Monitor-qt/tree/master

Regards.
Sandeep

sandeepcembed, 2024-08-20:
Thanks for the link,
My question is about does CDC virtual comport supports bus master directly? if not which one can i do in MCU side?
[image]
if you know which file we have to modify exactly or else can you specify any function name its helps me a lot.

rusoku, 2024-08-20:
Your driver must be able to work with a serialport.

sandeepcembed, 2024-08-20:
@rusoku I was able to send and receive CAN data over serial COM port using opensource QT-CAN-monitor source code as mentioned above. Which mean I can work with serial port over my NXP rt controller.

Now I wanted to communicate with my controller over the serial port using bus master. I gone through the shared code I didn't found where exactly port name or port opening is doing with desired speed?

S-adnan556, 2024-08-20:
@rusoku I have taken a copy of the Vector driver source code(Sources/BUSMASTER/CAN_Vector_XL) I wanted to know what changes I need to make so that I can select my hardware configure it and send and receive can data from busmaster UI.

I did check your driver link https://github.com/rusoku/busmaster_rusoku_driver but still not sure what exactly do i need to replace to make it work with my hardware

rusoku, 2024-08-20:
You need to rewrite all these functions to work with your hardware:

    HRESULT CAN_PerformInitOperations(void);
    HRESULT CAN_PerformClosureOperations(void);
    HRESULT CAN_GetTimeModeMapping(SYSTEMTIME& CurrSysTime, UINT64& TimeStamp, LARGE_INTEGER& QueryTickCount);
    HRESULT CAN_ListHwInterfaces(INTERFACE_HW_LIST& sSelHwInterface, INT& nCount, PSCONTROLLER_DETAILS);
    HRESULT CAN_SelectHwInterface(const INTERFACE_HW_LIST& sSelHwInterface, INT nCount);
    HRESULT CAN_DeselectHwInterface(void);
    HRESULT CAN_SetConfigData(PSCONTROLLER_DETAILS ConfigFile, int Length);
    HRESULT CAN_StartHardware(void);
    HRESULT CAN_StopHardware(void);
    HRESULT CAN_GetCurrStatus(STATUSMSG& StatusData);
    HRESULT CAN_GetTxMsgBuffer(BYTE*& pouFlxTxMsgBuffer);
    HRESULT CAN_SendMsg(DWORD dwClientID, const STCAN_MSG& sCanTxMsg);
    HRESULT CAN_GetBusConfigInfo(BYTE* BusInfo);
    HRESULT CAN_GetLastErrorString(std::string& acErrorStr);
    HRESULT CAN_GetControllerParams(LONG& lParam, UINT nChannel, ECONTR_PARAM eContrParam);
    HRESULT CAN_SetControllerParams(int nValue, ECONTR_PARAM eContrparam);
    HRESULT CAN_GetErrorCount(SERROR_CNT& sErrorCnt, UINT nChannel, ECONTR_PARAM eContrParam);
    HRESULT CAN_SetAppParams(HWND hWndOwner);
    HRESULT CAN_ManageMsgBuf(BYTE bAction, DWORD ClientID, CBaseCANBufFSE* pBufObj);
    HRESULT CAN_RegisterClient(BOOL bRegister, DWORD& ClientID, char* pacClientName);
    HRESULT CAN_GetCntrlStatus(const HANDLE& hEvent, UINT& unCntrlStatus);
    HRESULT CAN_LoadDriverLibrary(void);
    HRESULT CAN_UnloadDriverLibrary(void);
    HRESULT CAN_SetHardwareChannel(PSCONTROLLER_DETAILS, DWORD dwDriverId, bool bIsHardwareListed, unsigned int unChannelCount);

rusoku, 2024-08-20:
Take a look at CAN_VSCOM driver source code.It also uses a serial port interface with its hardware.

S-adnan556, 2024-08-21:
I have been examining the CAN_VSCOM driver source code and observed that it utilizes a statically linked library to facilitate serial communication operations, specifically for opening, closing, reading, and writing to COM ports.

S-adnan556, 2024-09-05:
I successfully transmitted CAN data from busmaster to my hardware and want to find where the transmission interval for the CAN messages is being set. On the receiving end, the time intervals is almost double of whatever is configured on the transmitting side.

When I manually send CAN messages from the COM port at specific intervals, I observe nearly the same intervals at the receiving end.

S-adnan556, 2024-09-25:
I resolved the issue of the cycle time being nearly double by dividing the nActualTimer by 2 (pouMsgItem->TxDetails.nActualTimer / 2).

For the COM port, I had used the VSCOM driver, modified it, and successfully transmitted and received CAN messages. Now, I'm considering switching to USB bulk mode. Should I use the same VSCOM driver with modifications, or is there another driver I can refer to as a better option?