rbei-etas/busmaster #1329: “Driver selection failed” problem

F001Man, 2025-07-20:
I have installed BUSMASTER v3.2.2 with TDM-GCC. When I click on driver selection it shows "Driver Selection Failed", but it worked well in two years before.
i tried in every driver, each one of driver selection is giving the same error.
also tried in my colleagues laptop, we all met this problem.
is there anyone else met this problem? 
thanks for your help!

gxurma, 2025-07-20:
you need to install all needed drivers that you need for your device.
Maybe it is only a Windows 11 problem?

Mit freundlichen Grüßen

Martin Gyurkó

https://www.engyneer.at (https://www.engyneer.at)
+43 680 1185 256

F001Man, 2025-07-21:
Dear Martin
Yes! I thought I can not use busmaster since we  upgrade the window11
for driver, I only use ETAS device, and I can use INCA very well.

Best regards,

Hanliang ZHANG

BEG engineering system China team 3 (BEG/ESY3-CN)
Bosch (China) Investment Ltd. | 455 Xinglong Road
Suzhou Industrial Park | Suzhou Jiangsu 215024 | P.R.CHINA
Tel. +86 21 2218-6784 | ***@***.******@***.***>
​

https://www.engyneer.at (https://www.engyneer.at)

GitBro-bit, 2025-08-27:
Hello, I encountered the same error: "Driver selection faliled" when i tried to connect. could you give me some hints on how to solve this problem.  The inca version is 7.4.4, busmaster version is 3.3.3 and the windows version is win11.
thanks a lot.

Zhang-Hanliang, 2025-08-27:
i still have the problem, and i think the problem is Win11.